wifi-tally/wifi-tally #8: we need a logo :)

czenker, 2020-05-10:


ruudboon, 2020-11-28:
Maybe we can work together on this. https://github.com/ruudboon/tally-blaster

czenker, 2020-11-28:
Hello and welcome.

> Maybe we can work together on this. https://github.com/ruudboon/tally-blaster

I'd love to see more people and ideas on board. You already have a logo and a case – so consider me envious and intrigued. :laughing: You also have some features implemented that I find quite interesting, like the dimming and a working Access Point configuration. So I am all up for joining forces to give the open source world a reliable and affordable tally light that is just amazing.

As on the logo:

I asked a friend of mine to create a logo, but he is just too busy the last months. The specs I gave him were

* monochrome, so it can be 3d-printed on a (potential) case
* small enough so it can be recognized when used as a browser favicon (32x32 px²)

If you have ideas I am all up for ideas.

ruudboon, 2020-11-28:
@czenker I got a big production this weekend. But let's have a chat somewhere next week to see if our ideas align. Cheers!

czenker, 2020-11-28:
@ruudboon I added an email to my profile (or just the one from the commits) and we will find a place to chat.

Brunsben, 2021-03-12:
I'm not a great graphic artist, but maybe that's an inspiration
[image]

MrTomek, 2021-03-12:
@czenker  Hi guys, I am willing to add something of my own, once I did a tally with HC-12 transmitters (433Mhz) because they have a greater range than

ruudboon, 2021-04-04:
[image]

czenker, 2021-05-15:
I felt that the project deserves something better than the generic "WiFi Tally" name along with the logo. So, say Hi to `vTally`:

[image]

ruudboon, 2021-05-15:
Can you make the new logo available as .svg (vector) for 3d printing?

czenker, 2021-05-15:
> Can you make the new logo available as .svg (vector) for 3d printing?

https://github.com/wifi-tally/wifi-tally/blob/stream-0.5/art/vtally-logo.svg

Will be on main as soon as `v0.5.0` is merged.

closed 2021-12-28 by czenker