wifi-tally/wifi-tally #9: consider using pkg to distribute hub

czenker, 2020-05-10:
Starting something on CLI is probably not what most practioners consider "easy". 

[Pkg](https://www.npmjs.com/package/pkg) allows to distribute the application with a node.js environment, reducing compatibility issues. The only downside is, that it hides the source files which could hinder participation.

czenker, 2020-08-26:
I played around with `pkg`. There is a nice post by Michael Hsu on [how to put a next.js application in pkg](https://medium.com/@evenchange4/deploy-a-commercial-next-js-application-with-pkg-and-docker-5c73d4af2ee). Still I ran into some issues:

1) Some files are not added and have to be manually selected. Ther are also some file path missmatches.
````
Using mixer configuration "null"
No video mixer connected.
Listening for Tallies on 0.0.0.0:7411
event - compiled successfully
Web Server available on http://localhost:3000
event - build page: /
wait  - compiling...
event - compiled successfully
Error: Cannot find module '../../../home/me/projects/wifi-tally/hub/node_modules/next/dist/next-server/lib/constants.js'
Require stack:
- /home/me/projects/wifi-tally/hub/.next/server/pages/_document.js
- /snapshot/hub/node_modules/next/dist/next-server/server/require.js
- /snapshot/hub/node_modules/next/dist/next-server/server/load-components.js
- /snapshot/hub/node_modules/next/dist/next-server/server/api-utils.js
- /snapshot/hub/node_modules/next/dist/next-server/server/next-server.js
- /snapshot/hub/node_modules/next/dist/server/next.js
- /snapshot/hub/server.js
1) If you want to compile the package/file into executable, please pay attention to compilation warnings and specify a literal in 'require' call. 2) If you don't want to compile the package/file into executable and want to 'require' it from filesystem (likely plugin), specify an absolute path in 'require' call using process.cwd() or process.execPath.
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:795:15)
    at Function.Module._resolveFilename (pkg/prelude/bootstrap.js:1350:46)
    at Function.Module._load (internal/modules/cjs/loader.js:688:27)
    at Module.require (internal/modules/cjs/loader.js:850:19)
    at Module.require (pkg/prelude/bootstrap.js:1229:31)
    at require (internal/modules/cjs/helpers.js:74:18)
    at eval (webpack-internal:///../next-server/lib/constants:1:18)
    at Object.../next-server/lib/constants (/home/me/projects/wifi-tally/hub/.next/server/pages/_document.js:103:1)
    at __webpack_require__ (/home/me/projects/wifi-tally/hub/.next/server/pages/_document.js:23:31)
    at eval (webpack-internal:///./node_modules/next/dist/pages/_document.js:20:18) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/me/projects/wifi-tally/hub/.next/server/pages/_document.js',
    '/snapshot/hub/node_modules/next/dist/next-server/server/require.js',
    '/snapshot/hub/node_modules/next/dist/next-server/server/load-components.js',
    '/snapshot/hub/node_modules/next/dist/next-server/server/api-utils.js',
    '/snapshot/hub/node_modules/next/dist/next-server/server/next-server.js',
    '/snapshot/hub/node_modules/next/dist/server/next.js',
    '/snapshot/hub/server.js'
  ],
  pkg: true
}
````

2) The atem mixer is not working properly. It somehow forces the server.js to start over and over again leading to port binding issues. 
````
Using mixer configuration "atem"
Connecting to ATEM at 192.168.178.200:9910
Listening for Tallies on 0.0.0.0:7411
Using mixer configuration "atem"
Connecting to ATEM at 192.168.178.200:9910
server error: Error: bind EADDRINUSE 0.0.0.0:7411
    at dgram.js:321:20
    at processTicksAndRejections (internal/process/task_queues.js:81:21)
    at process.runNextTicks [as _tickCallback] (internal/process/task_queues.js:62:3)
    at Function.Module.runMain (pkg/prelude/bootstrap.js:1380:13)
    at MessagePort.<anonymous> (internal/main/worker_thread.js:139:25)
    at MessagePort.emit (events.js:210:5)
    at MessagePort.onmessage (internal/worker/io.js:70:8)
starting
Using mixer configuration "atem"
````

So there is quite some work left and I am not really convinced to trust pkg as a stable way of distributing the hub. :frowning_face:

ruudboon, 2021-03-11:
Would it be an idea to pack the hub as an electron application so people can just download an executable and run it? Companion is doing something similar. A small tray application that shows the application (frontend + backend) is running and a link to open the browser.

czenker, 2021-03-11:
Interesting point. I had a look at [electron forge](https://www.electronforge.io/) just last week. 

If this works, I would like it. Especially with the addition of a tray icon. The thing that makes it very difficult for electron (pkg and nexe as well) is shipping [native addons](https://github.com/nodejs/node-gyp). And I don't think there is a way around them when adding a possibility to flash the Tally via USB or a native integration of Wirecast. There seem to be solutions around that, but for me – without any electron knowledge – it would take some time to figure out how to do it.

If you are willing to help there I would be very happy. You have some experience with electron.

ruudboon, 2021-03-11:
@czenker Sure. I can work on a proof of concept to see if this works. I did already a lot of work in my own project so we can copy some stuff from there. Maybe you can only help me explaining the current build process.

ruudboon, 2021-04-04:
I did some work in my [cloned repo](https://github.com/ruudboon/wifi-tally/tree/electron). What I mainly did is adding electron and using the build as it currently is. 
The application is starting express so the process will be killed when the application is closed. Currently I only tested it on mac but you can [grap](https://github.com/ruudboon/wifi-tally/actions/runs/717061864) the artifacts from the latest GitHub actions.

Before dropping a pull I would like to do the following.
* [ ] Create a basic layout that shows the current host ip
* [ ] Make it a tray icon app
* [ ] Improve CI flow for GitHub Actions

Future wishes for me
- Sign the application (I can maybe sponsor the Mac edition or even add it to the App Store)
- Basic status of connected mixer and tally's
- Update notification

ruudboon, 2021-04-18:
@czenker If you have time, could you check if above approach is something that fits your project? If yes, I would like to improve it and drop a pull request so you handle it yourself from there.

czenker, 2021-04-20:
@ruudboon I promise to have it done after the weekend. :crossed_fingers: If not, you know where to find me.

ruudboon, 2021-04-20:
@czenker No rush. Won't be able to work on it this week :D

closed 2021-12-28 by czenker