wifi-tally/wifi-tally #12: support Vmix

czenker, 2020-05-12:
https://www.vmix.com

lainol, 2020-06-05:
love to see this feature!! thank you very much, great work!

czenker, 2020-07-21:
Hey @lainol .

I added support for vMix today. It seems to work rather stable and looks good to me. Do you have a setup where you can give it a try in a real-world scenario?

lainol, 2020-07-22:
Yeah, tomorrow I can try It and leave feedback, thank you so much!

DVEntertainment, 2020-10-14:
I used it in a real world live stream event!
Someone built these tallys and i got to use them, they work great!
I'm planning on writing some buildguides for this. Is there any chance you would post or reference my 3dprint files for the boxes?

If anyone wants to buy pre-printed boxes you can comment on this post ofc (this is not the right place for that i know)

best regards
Daniël

czenker, 2020-10-14:
Hi Daniël

Thanks for confirming that it works in a real life scenario. Now I have peace of mind to close this issue. :smile: 

About the build guide and 3d prints: Of course! I love to see what people have built and happy to showcase them. And especially a 3d printable case is missing at the moment. So the more options there are – the better.

closed 2020-10-14 by czenker