wifi-tally/wifi-tally #13: support Wirecast

czenker, 2020-05-12:
https://www.telestream.net/wirecast/overview.htm

marianputsko, 2020-11-16:
Pls, integrtate to WireCast. Thank you

pistikz, 2020-11-16:
> Pls, integrtate to WireCast. Thank you

Could you make it to Wirecast? Thanx

marianputsko, 2020-11-17:
Si poklad 😆 pridal si aj do sekcie obs? Ci nejdeš už používať obs? Ty máš pro, nebudeš musieť. Na titulky si vyskúšaj toto https://heretorecord.com/graphics/

Prajem pekny den, s pozdravom

Marian Putsko, putsko.com, s.r.o.
Zapotocna 959/45
02801 Trstena, Slovakia

Tel. +421905879107
Mail: marian@putsko.com 
------—-----------------------------
www.putsko.com
www.liveproduction.sk
www.selfiebudka.sk
www.knihahosti.sk

czenker, 2021-02-05:
Supporting Wirecast is tricky.  They do not have any network based API, but only OLE objects. 

Long story short: I wrote a small Windows application that exposes parts of this API to the network. This application [WirecastTallyBridge can be found here](https://github.com/wifi-tally/wirecast-tally-bridge-win/releases/tag/0eeb46e2). This application is _highly experimental_ and I can't recommend to rely on it in production, but I would highly appreciate feedback if it works for you and how you managed to crash it.

The release of Wifi Tally with support for Wirecast can be found here. https://wifi-tally-nightly.s3.eu-central-1.amazonaws.com/wifi-tally-v0.4.0-1-g2ce945e.tar.gz

Please note that there is no version for Mac. Sorry. I only own bio-degradable apples, so help is highly appreciated.

### Steps

1. Download and run [WirecastTallyBridge](https://github.com/wifi-tally/wirecast-tally-bridge-win/releases/tag/0eeb46e2) on the same machine that Wirecast runs on
1. if Wirecast is not running WirecastTallyBridge should start it. Make sure to select a document.
1. Download and run [the dev version of Wifi Tally](https://wifi-tally-nightly.s3.eu-central-1.amazonaws.com/wifi-tally-v0.4.0-1-g2ce945e.tar.gz) and select Wirecast as Video Mixer. Enter the IP Wirecast is running (or `127.0.0.1` when on the same machine). There is no need to change the port number `1234`.

BotaRacha, 2021-03-09:
Hello everyone. Sorry but I cant download the wifi tally version with wirecast suport

BotaRacha, 2021-03-09:
> > Hello everyone. Sorry but I cant download the wifi tally version with wirecast suport
> 
> Thanks for the pointer. Nightly builds are only stored for 14 days. I regenerated the file, so it should be available again – for 14 days.

Thank you very much