wifi-tally/wifi-tally #18: Hub failing to launch

mlcook19, 2020-06-27:
I am having this error getting the hub to run. Any ideas how to solve it?

[image]

czenker, 2020-06-29:
I just built the current master branch as [v0.1-alpha3](https://github.com/wifi-tally/wifi-tally/releases/tag/v0.1-alpha3). So now the releases and the documentation should match up again.

This version should not show this problem when running production mode, but it now appears in dev mode – so this still needs fixing. But I hope this helps you in the mean time!?

[Background about the error](https://stackoverflow.com/questions/11928013/node-env-is-not-recognized-as-an-internal-or-external-command-operable-comman). Best fix seems to be to [use `cross-env`](https://stackoverflow.com/a/37131308).

czenker, 2020-07-20:
I just pushed 35a3714. Can someone verify the issue is gone for good?

kubikrubik21, 2020-07-21:
Confirmation

Hub works fine for me.

Tested v0.1-alpha3 and v0.1-alpha4 on Windows10 64bit

czenker, 2020-08-26:
This still does not seem to work without running `npm install` before.

czenker, 2020-08-26:
submitted a different fix

closed 2020-08-26 by czenker