wifi-tally/wifi-tally #22: Error: The Tally slowly blinks blue for an extended amount of time

gumorassi, 2020-07-14:
Hi, I have a problem and I can't solve it. I loaded all the files correctly, but Tally slowly flashes blue over a long period of time and doesn't stop. I tested it with several different wifi networks and it doesn't work. What could be happening? can you help me? @czenker 

Here's what appears on ESPlorer:

NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: master
	commit: 71a182caa7841cbb478ed90ede526dc881943c80
	release: 
	release DTS: 202001061321
	SSL: false
	build type: float
	LFS: 0x15000
	modules: encoder,file,gpio,net,node,pwm2,struct,tmr,uart,wifi
 build 2020-02-06 20:31 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
[INFO]  booted because of EXTERNAL_RESET
[INFO]  Connect to WiFi Rede_Morassi
E:M 1200
E:M 1200
> 

Thank you all.

czenker, 2020-07-14:
Hey @gumorassi 

Sure. We will get this figured out.

The `E:M` line points to an issue where there is insufficient memory left during run time. So to strike out a few options:

**Which files did you upload?**
You should use the `.lc` ones from the release package. When uploading the `.lua` files the NodeMCU has to compile them itself which uses up memory and could lead to this error.

Another idea - though very unlikely - is that the memory of your NodeMCU is smaller than it should be.
Remove all files from the NodeMCU, reboot it, open a terminal on the NodeMCU and run
````lua 
print(node.heap())
````
It should print something around
````
42648
````
If your number is significantly lower, this might be an issue.

gumorassi, 2020-07-14:
Hey @czenker First of all, thanks for the help.

1st test:
I removed all files, restarted and ran the

print (node.heap ())

Which printed the value of:

42200

2nd test:
I loaded the `.lc` files, before loading the `.lua` files, and the error was different this time:

NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: master
	commit: 71a182caa7841cbb478ed90ede526dc881943c80
	release: 
	release DTS: 202001061321
	SSL: false
	build type: float
	LFS: 0x15000
	modules: encoder,file,gpio,net,node,pwm2,struct,tmr,uart,wifi
 build 2020-02-06 20:31 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
lua: my-app.lc: bad constant in precompiled chunk
stack traceback:
	[C]: in function 'dofile'
	init.lua:1: in main chunk
	[C]: ?
>

czenker, 2020-07-14:
````
lua: my-app.lc: bad constant in precompiled chunk
````

That's not the most helpful error message I have ever seen. :smile: But I gave it a try and there seems to be something wrong with the distributed `lc` files. I created bug #24 for it as this is obviously _my_ fault.

But coming back to your issue. Ignore the releases for now.

To narrow the problem down further:

* I assume you are using Windows? And "NodeMCU PyFlasher" and ESPlorer as your tools of choice?
* Who is the vendor of the board you are using?
* Can you try a different board? Maybe even by a different vendor? (to rule out a hardware defect)
* Can/Did you try erasing the firmware and flashing from scratch?
* I guess you already smashed the `Reset` button multiple times and the problem persisted? (this should free all memory that could still be blocked by compiling the `lua` files on NodeMCU)

The good news is: I managed to flash something with those tools on Windows that shows symptoms similar to yours (getting stuck and out-of-memory errors). I have no idea if it is caused by the same thing, but it is something I can investigate.

gumorassi, 2020-07-15:
Got it, I think it may be this my-app.lc file that is corrupted.

* Yes, I am using Windows, NodeMCU PyFlasher and ESPlorer. Do you have any better suggestions?

* The vendor of the board is: AMICA

* I have another board from the same vendor and I will test to rule out the possibility of hardware defects.

* Yes, I used "flash_download_tool_v3.8.5" to clean the flash.

* Yes

* I downloaded the package "wifi-tally-v0.1-alpha3.zip" and followed the procedure at" https://wifi-tally.github.io/ "(I installed the firmware first and then added all the .lc files then the .lua and lastly the tally-settings).

I will try "wifi-tally-v0.1-alpha2.zip" to see if I have any success, what do you think?

gumorassi, 2020-07-15:
Sorry for my ignorance in passing the problems to you, maybe in some way that I can't understand very well, I have an initial knowledge, but I believe that together we will succeed.

czenker, 2020-07-20:
Sorry it took me some time to look into the issue with the broken `lc` files in the release.

It is fixed now, so would you mind giving [v0.1-alpha4](https://github.com/wifi-tally/wifi-tally/releases/tag/v0.1-alpha4) a try?

It works for me - and hopefully it does work for you too. There is no need to update the firmware - it has not changed in the release.

kubikrubik21, 2020-07-21:
I was running into the same problem with v0.1-alpha3.

Updated to v0.1-alpha4 yesterday and everything works fine for me now.

gumorassi, 2020-07-23:
@czenker 
show, it worked perfectly, congratulations and thank you very much.

closed 2020-07-23 by gumorassi