wifi-tally/wifi-tally #23: Build a case

czenker, 2020-07-14:
Some bare electronics on a breadboard are certainly appealing. But building a nice case around it that can be mounted on a hot shoe also has its benefits.

kubikrubik21, 2020-08-06:
My first tally prototype is ready!

I used a Wemos D1 mini as NodeMCU and built a [case from ebay](https://www.ebay.de/itm/VBESTLIFE-16-Kanal-Drahtloser-Wireless-Flash-Trigger-Set-Blitzausloser-Sender-DE/352991513180?hash=item522fef0a5c:g:He0AAOSwG8NeXxfC) around it. It needed some modifications, but now is a perfect fit.

Have a look
[image]
[image]
[image]

czenker, 2020-08-08:
Wow. This looks amazing!

AlexCruz02, 2021-04-07:
What Is the name of that big led.

AlexCruz02, 2021-04-07:
or the listing to buy it

ruudboon, 2021-04-07:
For anybody who's interested. I'm working on a case that I'm planning to share soon. I used these [leds](https://nl.aliexpress.com/item/32732047818.html?spm=a2g0s.9042311.0.0.42d44c4dGIz4Ce). I will add an USB-B or USB-A chassis because I broke too many usb micro connectors on the NodeMCU.  Besides that it has an GoPro mount that can be easily combined with for example [this](https://nl.aliexpress.com/item/32904947724.html?spm=a2g0s.9042311.0.0.7d784c4dkSdaEB) or [this](https://nl.aliexpress.com/item/4001244560045.html?spm=a2g0s.9042311.0.0.7d784c4dkSdaEB). Waiting for some deliveries to finish this.

[image]
[image]
[image]
[image]

sisensee, 2021-04-20:
> For anybody who's interested. I'm working on a case that I'm planning to share soon. I used these [leds](https://nl.aliexpress.com/item/32732047818.html?spm=a2g0s.9042311.0.0.42d44c4dGIz4Ce). I will [...]

This is pretty fantastic!

kubikrubik21, 2021-04-21:
> or the listing to buy it

CQL-410 RGB  / WK

https://www.tme.eu/de/details/cql-410rgb_wk/led-dioden-tht-20mm/polam-elta/cql-410-rgb-wk/

Amenasme, 2021-05-08:
Hi guys, I made this case with a 3d printer, it has room for 3 ws2812b leds for stage and 1 rgb led for operator. Then I diffuse the light by covering it with a transparent resin and some drop of white color (made for resin) . This resin will be colored entirely by the leds so we will see a full colored rectangle, and not only a small light. An hot shoe mount is attached on the bottom side and the usb is accessible on the sideso it can be powered by a power bank. I do not insert a battery inside the case. I prefer them small and lightweight. I power them with cameras usb or small powerbank.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

AlexCruz02, 2021-06-29:
Files?

AlexCruz02, 2021-07-24:
> Hi guys, I made this case with a 3d printer, it has room for 3 ws2812b leds for stage and 1 rgb led for operator. Then I diffuse the light by covering it with a transparent resin and some drop of [...]

You Can Share the files your 3d printed case

AlexCruz02, 2021-07-24:
puedes subir los archivos para poder imprimirlos