wifi-tally/wifi-tally #27: support OBS

czenker, 2020-08-08:
https://github.com/obsproject/obs-studio

DVEntertainment, 2020-11-10:
Yes please!

marianputsko, 2020-11-16:
Yes, please

czenker, 2020-11-20:
Your voices have been heared. :smile: 

I have added OBS support in the recent commits. It requires the [obs websocket plugin](https://github.com/Palakis/obs-websocket) and allows to track all _scenes_ with a tally light.

Could some of you give it a thorough testing with some real-life setup? A nightly build is available here:

  * https://wifi-tally-nightly.s3.eu-central-1.amazonaws.com/wifi-tally-v0.1.0-21-g389a298.zip
  * https://wifi-tally-nightly.s3.eu-central-1.amazonaws.com/wifi-tally-v0.1.0-21-g389a298.tar.gz
  * https://wifi-tally-nightly.s3.eu-central-1.amazonaws.com/wifi-tally-v0.1.0-21-g389a298.tar.xz

As I have not worked with OBS in production, maybe someone can answer me, if it is enough to only allow to select _scenes_  for the tally lights. It allows for Picture In Picture by embedding scenes into one another. But I am not sure how this is typically set up.

[image]

Have fun and I hope to hear of your experiences. :movie_camera:

DVEntertainment, 2020-11-21:
oh my god you are awesome. I'll test it this weekend!

czenker, 2020-12-21:
@DVEntertainment Any news on this? Did it work out?

marianputsko, 2020-12-21:
Helo.

I stoppt to use OSB, now I use Wirecast. 

Thank you.

Prajem pekny den, s pozdravom

Marian Putsko, putsko.com, s.r.o.
Zapotocna 959/45
02801 Trstena, Slovakia

Tel. +421905879107
Mail: marian@putsko.com 
------—-----------------------------
www.putsko.com
www.liveproduction.sk
www.selfiebudka.sk
www.knihahosti.sk

DVEntertainment, 2020-12-22:
> 
> 
> @DVEntertainment Any news on this? Did it work out?

I have used it a couple of times, and it never failed on me! Great implementation!

czenker, 2020-12-23:
Great to hear! Thanks for testing it.

closed 2020-12-23 by czenker