wifi-tally/wifi-tally #29: add support for WS2812b aka Neopixel LED Strips

flow1986, 2020-08-11:
Would it be possible to add support for the WS2812b LED Strips?
(also known as Adafruit Neopixel)

they are 5 Volt and you can connect them easy to the nodeMCU board with just 3 Pins:
GND, V, Signal...

thanks
Flow

czenker, 2020-08-11:
Hey Flow.

I like the idea. This is also an elegant solution to the problem that we might be draining too much current through the pins.

The only problem I see is that [some boards do not connect the 5V pin to the USB plug](https://arduino.stackexchange.com/a/51878). But I like this as an option nevertheless.

mlcook19, 2020-08-11:
I think this could be a good addition to the platform since 5v LED strip is
getting harder and harder to find.

czenker, 2020-08-28:
Hey @flow1986 and @mlcook19.

The latest [release 0.1.0](https://github.com/wifi-tally/wifi-tally/releases/tag/v0.1.0) has support for WS2812, if you want to check it out. 

I really like it and the strip is sooo much brighter than the LED strip I was using before :sunglasses: – so I will definitely go with this.

closed 2020-08-28 by czenker