wifi-tally/wifi-tally #35: Wemos D1 Mini

gumorassi, 2020-08-31:
Hello @czenker 
Is it possible to use a wemos D1 mini?
The curiosity arose for being a smaller size board.

czenker, 2020-08-31:
Yes, it works. 

@kubikrubik21 uses it in [his build](https://github.com/wifi-tally/wifi-tally/issues/23#issuecomment-669936670) and all the necessary pins are available. I also just gave the new NeoPixel / WS2812 support a try at home. So you should not have any trouble.

gumorassi, 2020-09-01:
I had a problem loading the firmware, the following message appears on the NodeMCU PyFlasher and it does not go away: "Uploading stub ..."
 What could it be?

czenker, 2020-09-01:
There could be many reasons. :thinking: 

First up: What's your experience with microcontrollers? I hope I am not getting too basic here, but these are the things I would consider first:

* it's perfectly normal for the process to take 1-2 minutes. But I don't assume _this_ is your problem.
* Do you use a baud rate of `115200`? Having it lower will increase the upload time significantly. Having it higher could lead to problems.
* Did you install the [driver for the USB-to-Serial chip](https://docs.thingpulse.com/how-tos/install-drivers/)? For the Wemos D1 mini this should be the CH840G.
* Does it behave differently on a different USB port?
* Do you have a different D1 Mini that you can try to rule out hardware defects? Or is it possible to flash a different firmware?
* Disconnect the D1 Mini from any circuitry. Sometimes it can interfere with the upload process.

If nothing helps, please post the full log of PyFlasher here.

closed 2020-10-05 by gumorassi