wifi-tally/wifi-tally #39: HOW TO SET UP THE HUB

AlexCruz02, 2020-09-27:
**im new to this and idont know how to set up the hub.******

czenker, 2020-09-29:
Hey @AlexCruz02 

Welcome on Github.

How to install the hub is described in [the Getting Started Guide](https://wifi-tally.github.io/getting-started.html#setting-up-the-hub). Where are you stuck or what does not work?

AlexCruz02, 2020-09-29:
hi my problem was solved but im having problems uploading the code to the nodemcu im using ESPlorer, and its giving this error
`HÈ110!„ˆëˆJéÿwmou¯{umowu¯»momw¯ou}»­»íëWaiting answer from ESP - Timeout reached. Command aborted.`

czenker, 2020-09-29:
Hm. I don't have an ESPlorer installation at hand at the moment. According to the internet the error message poped up again recently. It could be related to the new NodeMCU Release that has not been integrated, but some [people report that clicking the FS Info button](https://github.com/nodemcu/nodemcu-firmware/issues/784#issuecomment-613120211) makes the error go away. Might be worth a try. 

If it does not help you could start with a simple Hello World Application to check that the communication between ESPlorer and your microcontroller work. Here are a few examples:

* https://www.electronicwings.com/nodemcu/getting-started-with-nodemcu-using-esplorer
* https://odd-one-out.serek.eu/projects/esp8266-nodemcu-getting-started-hello-world/

czenker, 2020-11-11:
Closed, because the discussion went stale.

closed 2020-11-11 by czenker