wifi-tally/wifi-tally #41: How to setup node.js & hub on Mac OS?

pixelfixel, 2020-10-08:
Unfortunately I'm not a node expert and can't manage to start the hub on my Mac. I have installed the node.js packages and started them by double clicking on "node". This opens a new terminal window.

What should I do now? Where do I enter the hub directory and how do I start it? What is meant by "npm run start"?

I would be very grateful for a more detailed explanation!

Thanks a lot!
Stefan

czenker, 2020-10-10:
Welcome Stefan.

You need to [open a good old terminal](https://www.wikihow.com/Open-a-Terminal-Window-in-Mac), change to the directory where you unpacked the downloaded files (like `cd /your/path/to/wifi-tally-v0.1.0/hub`) and run the command there `npm run start`.

I don't know what starting the `node` application does – but whatever it does, I am pretty sure it is not necessary to get the hub running. :wink:

czenker, 2020-11-11:
Closed because discussion went stale.

closed 2020-11-11 by czenker