wifi-tally/wifi-tally #42: How to fix error loading module 'my-wifi' from file 'my-wifi.lc'

PeterThtr, 2020-11-10:
Hello everybody!
I would love get this smart project running and use wifi-tally for our live streaming events.
I made to flash the firmware to my ESP8266 and upload all necessary files (using ESPlorer). But during startup of the ESP8266 (new NodeMcu V3) I always get this "error loading module 'my-wifi'" message. Can somebody please help me fixing this?

Here is the output:

`NodeMCU 3.0.0.0 
	branch: HEAD
	commit: 8d091c476edf6ae2977a5f2a74bf5824d07d6183
	release: 3.0-master_20200610
	release DTS: 202006092026
	SSL: false
	build type: float
	LFS: 0x15000 bytes total capacity
	modules: encoder,file,gpio,net,node,pwm2,struct,tmr,uart,wifi,ws2812
 build 2020-08-26 14:11 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
error loading module 'my-wifi' from file 'my-wifi.lc':
	my-wifi.lc: bad integer in precompiled chunk
stack traceback:
	[C]: ?
	[C]: in function 'require'
	src/my-app.lua:7: in main chunk
	[C]: in function 'dofile'
	init.lua:1: in main chunk
	[C]: ?
> `

Thanks a lot!

Peter

czenker, 2020-11-11:
Hi Peter and welcome.

The error usually points to the problem that the firmware and the version the `lc` files where built with are different. 

I assume you used the firmware and the `lc` files from the [latest release](https://github.com/wifi-tally/wifi-tally/releases/tag/v0.1.0). Right? I just gave it a try and it works perfectly for me (on Linux).

What you could try is to download the [source code of the latest release](https://github.com/wifi-tally/wifi-tally/archive/v0.1.0.zip) and let ESPtool compile the included `lua` files. This would not fix the underlying issue, but might help you to continue with your project.

PeterThtr, 2020-11-11:
Hi Christian,

thank you for your quick answer. Yes, I used the lc files from the latest release. Unfortunately this is one level to high for me. I have python installed and the ESPtool up running (on my PC) but I don't know how to compile the lua files on my ESP8266. Is there a how-to you could recommend? 

Thanks a lot!

Peter

czenker, 2020-11-11:
Sorry. I wrote _ESPtool_ in my last post, but meant _ESPlorer_.

Don't fret: If you have the firmware on your ESP8266 it can compile lua files for you. ESPlorer has multiple buttons in the bottom left to compile `lua` files into `lc`. Just remember to compile all `lua` files, **except** `init.lua`. It has to be uploaded as is.

PeterThtr, 2020-11-12:
Hi! 
I have done it.
Thanks to your hints everything is working smoothly now. 
Great project. 
Thanks again and best regards!

czenker, 2020-11-14:
:bow:

closed 2020-11-14 by czenker