wifi-tally/wifi-tally #45: Support Roland VR-50HD MKII

ruudboon, 2020-12-04:
I would like to request for the Roland VR-50HD MKII

JWandscheer, 2021-04-06:
Sadly it's the third way of getting tally information from Roland devices except from midi and smart tally. It's possible via  sockets - there are already a few implementations out, seems not to be very complicated.
I'm not going to have a VR-50HD MKII mixer here in the near future, is it still necessary to implement support?

leusound, 2021-12-22:
I'm interested to get the VR-50HD MK2 in - how ist the status? Thanks :-)

czenker, 2021-12-29:
No progress as far as I know.

But if you have access to one, that might be a good start. It should be doable, but it is hard without a testing device. If you are willing to contribute or help, it could be possible.

leusound, 2021-12-29:
Hallo Christian
We have vr-50hy mk2, atem mini pro and vmix on our rental stock snd frequently in use. However we are no Software engineers (I’m electrical engineer) We would love to test and assist/support/controbute in a tally option for the vr-50hd mk2. We have on feb. 2nd a large production where we urgently need a tally at least for the canera-operator. Let me know what you think.

czenker, 2021-12-29:
Hey. That's great. I think, together we can get this thing working.

After reading the documentation and checking a reference implementation I might have a working prototype.

You can install it via npm ([see documentation](https://wifi-tally.github.io/download.html#run-from-npm)) using the following command:

````bash
npm install -g  https://wifi-tally-nightly.s3.eu-central-1.amazonaws.com/vtally-0.5.1-rc1-1-g23ce049/vtally-0.5.1-rc1-1-g23ce049.tgz
````

Contact me via my email address in my profile and I send you my personal contact information. So we can coordinate how to proceed.