wifi-tally/wifi-tally #50: Disable power status

ruudboon, 2021-02-06:
Currently the first led when using WS2812 is showing the status. (dimmed green when all good). 
Would it be possible to add an option to disable this and leave it all dark?

czenker, 2021-02-06:
Do you want to turn off the operator light completely or just disable the "everything ok, but you are neither on preview nor live"-light?

The first thing can be done with [`operator.ws2812 = 0`](https://github.com/wifi-tally/wifi-tally/blob/master/documentation/docs/tally.md#using-ws2812-neopixel-etc).

ruudboon, 2021-02-06:
Disable the everything ok, nothing switched green status led. When setting the operator led to low intensity the green ok is confused with the preview.

czenker, 2021-02-06:
Added it here: https://wifi-tally-nightly.s3.eu-central-1.amazonaws.com/wifi-tally-v0.4.0-2-ga2907b4.tar.gz

and will be in the next release.

ruudboon, 2021-02-08:
Thnx buddy

closed 2021-03-11 by ruudboon