wifi-tally/wifi-tally #51: Roland V8HD

Brunsben, 2021-02-10:
Hello,
is it possible to support the Roland V8HD too?

ruudboon, 2021-02-10:
I think it has the same protocol as https://github.com/wifi-tally/wifi-tally/issues/45 Will try to borrow a device soon.

Brunsben, 2021-02-22:
Would it help if we log the data traffic on the USB on our V8HD with Wireshark and make it available?

ruudboon, 2021-02-22:
Ah, I assumed it was network connected as well. You could check if a local network is created over the USB port but if that isn't the case it will be harder to implement this.

Brunsben, 2021-02-22:
I'll be back in the club tomorrow. Then I'll see if I can find out anything.

Brunsben, 2021-02-27:
Hey, it seems there is no local network.

Brunsben, 2021-03-06:
A friend told me that it is probably a midi signal.

czenker, 2021-03-10:
@Brunsben Thanks for looking into it. From what I see, an iPad app is the _only_ remote thing that you can connect to the mixer at the moment!?

I'm not expecting too much, but maybe contacting Roland with your problem and asking if they can supply you with a documentation of the serial protocol can help. Explaining that you are interested in implementing a solution, could help going forward. Worst case is that they don't answer. In the best case other open source projects, like Companion, benefit too.

On a side note: I'm currently working on a way to [flash the Tallies from within the Hub via USB](https://github.com/wifi-tally/wifi-tally/tree/flasher). So a serial library will be shipped with the next release anyway, which might help with integrating your Roland V8HD.

Brunsben, 2021-03-18:
> From what I see, an iPad app is the only remote thing that you can connect to the mixer at the moment!?

Yes.

I'll try on Friday with a friend to see if we can somehow get data out of there. He's just doing his master's degree in computer science, maybe he'll get something done.

JWandscheer, 2021-03-21:
Did id. It can be done via Midi, you have to send a SysEx DataRequest Message to the Mixer, it will return the tally status for the requested input (more information in [V-8HD manual](https://static.roland.com/assets/media/pdf/V-8HD_reference_eng03_W.pdf)).
In some other mixers, Roland has a SmartTally function, where the input status can be determined via simple http requests to the mixer (one request per channel needed). I've written a small Python script that basically simulates a SmartTally server, to save http-requests all statuses are always output. Additionally, I added an experimental support for requesting this dummy-server in wifi-tally.
The more elegant solution would be to connect the hub directly to the mixer via USB and read out the midi information directly in the wifi-tally, without going the diversions via another server-script. Think this can be done via [node-midi](node-midi). I will try this in the next few weeks.

czenker, 2021-03-22:
Great work, @JWandscheer! If you need help integrating it into wifi-tally, please ping me.

closed 2021-12-28 by czenker