wifi-tally/wifi-tally #52: RGB/GRB WS2812

DVEntertainment, 2021-03-06:
Hi Guys!

I have some GRB ws2812, and as far as i know, there is no way to switch this. This means that my preview is red, and my program is green. A simple switch in the hub would solve the problem!

Thanx!

czenker, 2021-03-09:
Hey.

I have added this as a feature to the `tally-settings.ini` (my thinking being, this is a feature of the tally light itself that you are never required to change if you don't change the hardware) Give it a try here:
https://wifi-tally-nightly.s3.eu-central-1.amazonaws.com/wifi-tally-v0.4.0-4-g103f657.tar.gz

and let me know if it works for you.

czenker, 2021-03-09:
Erm, sorry. Did not want to leave you hanging, but forget the mention I updated the [documentation](https://wifi-tally.github.io/tally.html) on how to do it as well :smile: 

`tally-settings.ini` should contain something like:

    operator.ws2812=1 rgb
    stage.ws2812=5 grb

DVEntertainment, 2021-03-10:
> 
> 
> Erm, sorry. Did not want to leave you hanging, but forget the mention I updated the [documentation](https://wifi-tally.github.io/tally.html) on how to do it as well 😄
> 
> `tally-settings.ini` should contain something like:
> 
> ```
> operator.ws2812=1 rgb
> stage.ws2812=5 grb
> ```

Yes This is exzactly where i was looking for! Thanx for the update of documentation!
Yes, tally-settings.ini is more than enough.

closed 2021-03-10 by czenker