wifi-tally/wifi-tally #56: Failed to start the hub on a Raspberry Pi 4

Brunsben, 2021-03-17:
Hi,

i tried to use wifi-tally on a Raspberry Pi 4.

When I execute the command "npm run start" in the Hub I get the following error:

`npm ERR! missing script: start`

When I run "npm install", I get the following message:
`pi@Wifi-Tally:~/wifi-tally/hub $ npm install

> core-js@2.6.12 postinstall /home/pi/wifi-tally/hub/node_modules/babel-runtime/node_modules/core-js
> node -e "try{require('./postinstall')}catch(e){}"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Collective or Patreon: 
> https://opencollective.com/core-js 
> https://www.patreon.com/zloirock 

Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)

> core-js@3.8.1 postinstall /home/pi/wifi-tally/hub/node_modules/core-js
> node -e "try{require('./postinstall')}catch(e){}"

> core-js-pure@3.8.1 postinstall /home/pi/wifi-tally/hub/node_modules/core-js-pure
> node -e "try{require('./postinstall')}catch(e){}"

> ejs@2.7.4 postinstall /home/pi/wifi-tally/hub/node_modules/ejs
> node ./postinstall.js

Thank you for installing EJS: built with the Jake JavaScript build tool (https://jakejs.com/)

> cypress@6.6.0 postinstall /home/pi/wifi-tally/hub/node_modules/cypress
> node index.js --exec install

Installing Cypress (version: 6.6.0)

  ✖  Downloading Cypress    
    → Cypress Version: 6.6.0
     Unzipping Cypress      
     Finishing Installation 
The Cypress App could not be downloaded.

Does your workplace require a proxy to be used to access the Internet? If so, you must configure the HTTP_PROXY environment variable before downloading Cypress. Read more: https://on.cypress.io/proxy-configuration

Otherwise, please check network connectivity and try again:

----------

URL: https://download.cypress.io/desktop/6.6.0?platform=linux&arch=ia32
Error: Failed downloading the Cypress binary.
Response code: 404
Response message: Not Found

----------

Platform: linux (Debian - 10)
Cypress Version: 6.6.0
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules/webpack-dev-server/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules/watchpack-chokidar2/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.2.1 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.2.1: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.3 (node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! cypress@6.6.0 postinstall: `node index.js --exec install`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the cypress@6.6.0 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-03-17T20_24_32_307Z-debug.log`

What is the reason for this and how do I get it going again? 
I don't use a proxy.

greetings

czenker, 2021-03-18:
Hey Ben.

I have not tried to run the hub on a Raspberry Pi. And the download packages are build for the amd64 architecture. So the Raspi is not really supported at the moment. You should get it to run, though. Execute `npm ci` once and then run `npm run start` again. 

If this does not help, please let me know which version you are running or post the content of the `hub/package.json` file. Also: You might have downloaded the Source Code, not the [actual release](https://github.com/wifi-tally/wifi-tally/releases/download/v0.4.1/wifi-tally-v0.4.1.zip)?

Brunsben, 2021-03-18:
Hi,
I got the repository via "git clone https://github.com/wifi-tally/wifi-tally/".

When I run npm ci (or npm install) I always get this error:
`Cypress cannot write to the cache directory due to file permissions`
and then a few more errors because the Cypress directory cannot be accessed.
If I then execute "npm run start" I get the error again
`npm ERR! missing script: start`

The content of the package.json is [this](https://pastebin.com/Ezt5VcP7)

czenker, 2021-03-18:
Ah. That explains the behavior. Please download the [latest release v0.4.1](https://github.com/wifi-tally/wifi-tally/releases/download/v0.4.1/wifi-tally-v0.4.1.zip) and use that instead. You might need to run `npm ci` there too, but apart from that the `start` command should work.

---

Running the hub from the cloned repo is only useful for development. It pulls more libraries that are not needed in production (like Cypress, Typescript, etc) and it boots up very slowly. You can read [hub/Readme.md](https://github.com/wifi-tally/wifi-tally/tree/master/hub#run-the-development-setup) if you are interested how to run it. But if you want to use it in production – the releases are way easier to set up and much faster.

Hope that helps.

Brunsben, 2021-03-18:
Ah, OK, that seems to work, Thanks

closed 2021-03-18 by Brunsben