wifi-tally/wifi-tally #57: Operator Light Brightness

ruudboon, 2021-04-03:
In theaters I noticed that 20% sometimes is even too bright.
Can we allow to go to 1%? Or to create consistency even go to off like the stage light?

czenker, 2021-04-08:
LEDs and their non-linear dim curve... :laughing: I don't see a problem with decreasing that value.

I would rather not allow to turn the operator light off completely. The Tally should be beginners friendly and turning the operator light off by accident might be very confusing, because the Tally might seem broken. On the other hand I do not see a reason why anyone would want to turn off the operator light completly. As a workaround... any production should have some Gaffer Tape, right? :wink:

ruudboon, 2021-04-08:
1% works for me. Thnx

closed 2021-05-16 by czenker