wifi-tally/wifi-tally #60: Not working with Vmix 24.0.0.56

tobiswdec, 2021-04-14:
Older Versions checked, but Vmix 24.0.0.56 isnt working. 

Sometimes the hub shows a video mixer connected, but you can not assing any inputs to tallylights.

Amenasme, 2021-04-29:
the same for me, it doesn't work. web tally doesn't connect

ruudboon, 2021-04-29:
I tried with 24.0.0.57 and didn't noticed any issues. Could you test with 24.0.0.57 as well?

czenker, 2021-04-29:
Also works for me on Win10 Build `19042.928` with NodeJS `14.16.1` and vMix `24.0.0.57`. 

The thing that looks strange in your config is the port `8088` used for vMix. Wifi-Tally uses the [TCP Api](https://www.vmix.com/help24/index.htm?TCPAPI.html) on port `8099` – unless you are doing some NATting or port forwarding there is usually no reason to change that. Port `8088` is an HTTP API by vMix that is not used by Wifi-Tally.

Amenasme, 2021-04-29:
Yeahh, now it works, I was using 8088 port too. With 8099 it works. Thank you! Maybe this thing must be written in the "getting started" section of the website to help people that don't know that. 

I can't connect to atem mini too, I know the ip address of the atem mini but I don't know the port. Where can I read what port the atem mini is using? Thank you

tobiswdec, 2021-04-30:
closed, it fixed my issue

closed 2021-04-30 by tobiswdec