wifi-tally/wifi-tally #61: no memory ESP

mnoscak, 2021-04-14:
Hi my ESP8266-12 4M after upload code (lua, lc) says:

modules: encoder,file,gpio,net,node,pwm2,struct,tmr,uart,wifi,ws2812
 build 2021-03-07 10:38 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
E:M 2155784
not enough memory
stack traceback:
	[C]: in function 'dofile'
	init.lua:1: in main chunk
	[C]: ?

Please why?     Thx Michal

czenker, 2021-04-20:
Hey Michal

That looks like a strange issue. 

How much RAM can be used by the application? It is easy to find out if you remove `init.lua` from the ESP and reboot. `print(node.heap())` tells you how much RAM is free for use by the application. It should be a value around `40000`. For me it is `43520`. If it is significantly less, then it seems the firmware takes up to much space or your ESP has less memory than usual.

Do you remember where you bought the ESP? I would be interested to get one in my hands if it does not work.

czenker, 2021-05-15:
Closed for inactivity. Unclear how to reproduce.

closed 2021-05-15 by czenker

Unyck, 2022-04-02:
I have the same issue, i check with the print(node.heap()) and my cheap have 42144 ram space, i got the exact same error than mnoscak the cheap come from Azdelivery its a nodeMcu V3