wifi-tally/wifi-tally #63: vMix: detect if stream or recording is active

czenker, 2021-04-29:
The TCP Api seems to [support detection if stream or recording is active](https://forums.vmix.com/posts/t24912-Expand-TCP-API-with-subscription-model-for-changes-to-XMLTEXT). We could use that to provide the same setting that OBS already has to limit when the `program` state is actually shown.

ruudboon, 2021-04-30:
Great feature but there are situations where obs or vmix isn't streaming but a different platform is. Can we add this as option?

czenker, 2021-05-12:
> Great feature but there are situations where obs or vmix isn't streaming but a different platform is. Can we add this as option?

You mean support for "chaining" video mixers that we talked about? Yes, I also have a setup that would benefit. I am a bit surprised, but it seems there is no issue for that yet.