wifi-tally/wifi-tally #65: uploading files to nodemcu

Amenasme, 2021-05-01:
Hi, i need some help to upload the file tally-setting.ini to the nodeMCU.
I am using ESPlorer and as you can see from the attached picture I got this error. I don't know the lua script, I just uploaded all the files and replace the ssid, password and ip on this file. Someone can help me? thank you
[image]

czenker, 2021-05-02:
Hey.

ESPlorer seems to *execute* `tally-settings.ini` as if it was LUA code. What you want to do is just move files around (ESPlorer can do sooooo much more than that) - **no** compiling or running needed. Also you need to rename `tally-settings.ini.example` to `tally-settings.ini` before uploading.

---

If you can't get it to work, you might give [NodeMCU-Tool](https://github.com/AndiDittrich/NodeMCU-Tool) a try too. I use that on Windows regularily and - personal taste - find it easier to use.

````bash
# install nodemcu-tool
> npm install nodemcu-tool -g

> nodemcu-tool devices
# the output looks similar to this:
#   [device]      ~ Device filter is active - only known NodeMCU devices (USB vendor-id) will be listed.
#   [device]      ~ Connected Devices | Total: 1
#   [device]      ~ - COM3 (Silicon Labs, USB\VID_10C4&PID_EA60\0001)

# Take note of the port of your connected device (COM3) in this example. 
# You need to substitude it in all following commands

# then upload all the files
> cd Downloads\wifi-tally-v0.4.2\tally
> nodemcu-tool upload *.lc init.lua tally-settings.ini -p COM3

# and reboot (you might as well push the RST button or replug power)
> nodemcu-tool reset -p COM3
````

Amenasme, 2021-05-02:
I was not renaming `tally-settings.ini.example` to `tally-settings.ini`. Now it is working very good! thank you

closed 2021-05-03 by Amenasme