wifi-tally/wifi-tally #66: "On Air" sign for Vmix

Amenasme, 2021-05-08:
Hi, is it possible to use wifi-tally technology to switch an "on air" sign automatically when vmix is recording or streaming? I tried for example to put a transparent input on overlay 4, so it is always on air (but transparent, so not visible) and a tally connected to it. Then I programmed that when I start recording, it automatically goes on overlay 4. Is it possible to make this in a simpler way, and without using an overlay channel? Maybe this feature can interest other people?

Amenasme, 2021-05-12:
@czenker can I ask your opinion about this?

czenker, 2021-05-12:
Hey @Amenasme

Sure. 😄 I like the idea of having an "On Air" sign functionality. Buying one of those cheap light boxes and turning it into a "We are live" sign sounds like something I would love to do. 😉 

I am just not sure what the best logic there is.

1. The easier version is to have a tally that shows the "on air" state if *any* channel of the mixer is "on air". I the dropdown could just show another option "any".
2. The flexibel version would be to allow multiple selections in the dropdown. But I am not sure if this is really needed.

Considering #63 is implemented, your use case would be solved with Option 1. Correct?

Option 2 would be better if you use lots of pre-recorded clips in vMix and your team could relax during that time. But this makes the UI more complex.

What are your preferences.

Amenasme, 2021-05-12:
I mean, the on air lights must be activated if recording or streaming in vmix are active, not connected to an input

Amenasme, 2021-05-12:
#63 I read this. If You can access to this data (vmix recording state and vmix streaming state) so maybe you can connect a tally to theese vmix states

Amenasme, 2021-05-12:
Someting like this, a sign that will switch on when vmix is recording or streaming

[image]

czenker, 2021-05-12:
> I mean, the on air lights must be activated if recording or streaming in vmix are active, not connected to an input

From the end-users perspective this should be the same, but it would be easier to implement this in a way that all other video mixers can profit as well. 

I might need to look into it for vMix, but I don't expect there would be a (real world) situation, where you are streaming, but **no** input is active. 

Did you build that Recording sign? It looks awesome. I need one of those, too. 😆

Amenasme, 2021-05-12:
Maybe I am streaming a pre-recorded video, or a powerpoint and in that case I want that all tally are off and recording/streaming sign is on. 

On the other side, I can have camera 1 in program but I am not streaming or recording yet, in rehersal for example, so I want that tally works but recording7streaming sign is off.

That is a picture taken from internet but I built 2 on air sign controlled by nodeMCU with a domotic app (Blink for android) and I can chang the light status sending web request to the server of the app. It works but it is difficult to set up, and little weird to use with vmix. I work in a medium television studio that wants theese sign so when they saw on air sign is on, they can't open doors or make noise.  I don't have picture of my signs now but I can take photo the next time I go there