wifi-tally/wifi-tally #67: Access to Build with Roland Support

joshuahornig, 2021-05-12:
Hello,

when will you release your new version with Roland V8HD Support ? 
if i run the Nightly i get the following error:

> wifi-tally@0.4.2-31-gbeb37f1 start /home/pi/wifi-tally/hub
> node server.js --env=production

/home/pi/wifi-tally/hub/node_modules/bindings/bindings.js:121
        throw e;
        ^

Error: /home/pi/wifi-tally/hub/node_modules/midi/build/Release/midi.node: wrong ELF class: ELFCLASS64
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1127:18)
    at Module.load (internal/modules/cjs/loader.js:933:32)
    at Function.Module._load (internal/modules/cjs/loader.js:774:14)
    at Module.require (internal/modules/cjs/loader.js:957:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at bindings (/home/pi/wifi-tally/hub/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/home/pi/wifi-tally/hub/node_modules/midi/midi.js:1:31)
    at Module._compile (internal/modules/cjs/loader.js:1068:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1097:10)
    at Module.load (internal/modules/cjs/loader.js:933:32)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! wifi-tally@0.4.2-31-gbeb37f1 start: `node server.js --env=production`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the wifi-tally@0.4.2-31-gbeb37f1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-05-12T16_02_38_579Z-debug.log

czenker, 2021-05-12:
Hi.

I hope to fix the issue on the coming weekend. Unfortunately it is not possible to ship the code the same way as before for all architectures.

But it should be easy to fix: Open a command line, go into the `hub/` directory, and execute `npm ci`. Then start again. 

As you are on a raspberry pi (?), you might need to install libasound2-dev before (`sudo apt-get install libasound2-dev`). It is needed to build the node extension for midi support.

I hope that helps.

joshuahornig, 2021-05-12:
Hi

thank you verry much, it it now running, i will test it with a real mixer on this weekend

closed 2021-05-12 by czenker