wifi-tally/wifi-tally #70: What's the connection?

pwright2b, 2021-05-13:
I realize this is a really dumb question ... but what is the connection between the ATEM and the Hub?  I don't find any description or reference to it.  And I don't find any schematics on the BlackMagic site to help.

czenker, 2021-05-13:
Hi. There are no stupid questions. 

I am not aware of any *official* documentation on the interface. It is an UDP event stream with binary data – so it is tricky to interpret. The interface is used officially by ATEM Software Control and has been [first reverse engineered and documented by SKAARHOJ](https://www.skaarhoj.com/discover/blackmagic-atem-switcher-protocol). This is also the only documentation of the interface I know of, but it is no longer maintained and might be outdated. We use [
tv-automation-atem-connection by NRK](https://github.com/nrkno/tv-automation-atem-connection) under the hood to communicate with ATEM. They have a working and maintained implementation, but – in their case – the source is the documentation.

pwright2b, 2021-05-13:
Well, but sometimes there are badly phrased ones.  I'm asking the really basic question of what kind of wire runs between the ATEM and the Hub.  USB?  9 Pin Serial?  VGA?  

Thanks for the quick reply.

czenker, 2021-05-13:
Yeah – I got this one wrong. Still not a stupid question though :smile: 

The answer is **ethernet**. There is a plug on the back labeled "Switcher Control" or "ATEM Control". Just connect that to a switch on your LAN and your LAN should give the ATEM Mixer an IP address. You might have to look the address up in your router, because I am not sure if your ATEM can show it to you. That is the IP you enter in the Hub. Just go with the default port and it should work.

For more details you can check out [this video on Youtube](https://youtu.be/7w3AGMPB83s?t=103) (this does not only work on ATEM Mini, but also Television Studio) or the section "Connecting to a Network" in the according handbooks.

And feel free to follow up if you need more detailed help.

pwright2b, 2021-05-14:
Fantastic.  Thank you for that.  And, yes, I may be back.  But I think I have enough to begin now.

closed 2021-05-15 by czenker