wifi-tally/wifi-tally #71: Me again.  How to install.

pwright2b, 2021-05-15:
OK.  Now I have an EeePC running Debian.  And I have downloaded the wifi-tally-v0.4.2.zip  and wifi-tally-v0.4.2.tar.gz.   But I have no idea how to/whether to install them.

----Paul-----

czenker, 2021-05-15:
Both of the archives contain the exact same files – so it does not matter which one you use. `tar.gz` is the format I would go with for linux, because any distribution can handle it without the need for additional packages. You just need to extract the archive using your file manager and should end up with a directory structure like that

````
wifi-tally-v0.4.2
  - documentation
  - firmware
  - hub
  - tally
````

You can ignore `firmware` and `tally` for now. You only need them when [building your own hardware](https://wifi-tally.github.io/getting-started/setup-wifi-tally.html). Open a terminal and enter the `hub` directory.

````bash
cd ~/Downloads/wifi-tally-v0.4.2/hub
````
(Your path might vary, depending where your stored it)

You might need to install nodejs, if you have not already:
````bash
sudo apt-get install nodejs
````

Then start the hub
````bash
npm run start
````

... and you can open a browser and should see the hub on http://127.0.0.1:3000

pwright2b, 2021-05-15:
Woo-hoo, it's  running.  Your instructions were great but I'm going to add a couple of things here because I expect some other noobs will find this thread useful.

Main guidance is:  when you get an error message, Google is your friend.

When I tried the sudo apt-get, I was informed I was not on the sudoers list.  Apparently a common fresh-install Debian issue.

Once I had nodejs installed and tried to run the hub, I was told npm was not a known command.  node -v worked but npm -v did not.  Again, apparently not all that uncommon.  A stackoverflow page gave me a curl command to run to fix it but then I had to install curl.

Finally , all seems to work.

Thank you.

-----Paul-----

pwright2b, 2021-05-23:
Aaand, I'm back.  Two pics attached from my present setup.  Which almost but not quite runs.
[image]
[image]

Asbestos I can tell, the wifi-tally and the hub are fully working.  They are all static IPs and connected by ethernet.  I have created one web tally on the hub and when I click 'Connect' I get a screen that turns colors depending on how Camera One is set.

My problem comes when trying to access from outside via wifi.  At first I attached a simple AP and tried to set up dhcp on the same machine as the hub.  Never could get configuration right.  All the config guides assume there will be an internet connection and trying to leave that out upsets them.

Finally I realized I had an AP that had its own dhcp server.  I now have that attached to the system.  It seems to work.  Some of my phone network utilities can see 10.55 and 10.240 within the network.  But I cannot find a way to log in to get the phone to acts as a wifi tallly.  I am sure I am missing something basic but hope that you can point it out to me.

I have to admit the 0.0.0.0:7411 notation has me stonkered.

-----Paul-----

czenker, 2021-05-24:
Hey.

Great you made it this far.

> Asbestos I can tell, the wifi-tally and the hub are fully working. 

Yes. Connection from Hub to ATEM looks perfect. :+1: 

> I have to admit the 0.0.0.0:7411 notation has me stonkered.

This address will not help you much. `0.0.0.0` is a special IP address that basically means that connections on all your network devices are accepted. So if your EeePC was connected via wire and wifi you could connect to the hub with either IP address. `7411` is the port.

To get the IP address of your EeePC, type `ip addr`. It returns something like this: 

````plain
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp4s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
...
    inet 192.168.178.42/24 brd 192.168.178.255 scope global dynamic noprefixroute enp4s0
       valid_lft 861173sec preferred_lft 861173sec
    inet6 fe80::6fb5:452e:c846:5c10/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
3: virbr0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether 52:54:00:28:27:7d brd ff:ff:ff:ff:ff:ff
    inet 192.168.122.1/24 brd 192.168.122.255 scope global virbr0
       valid_lft forever preferred_lft forever
...
````

Look for the address starting with `192.168.10` – that is the address you want your smartphones to connect to. In my example this address is `192.168.178.42`. So I type `http://192.68.178.42:3000` in my browser.

> My problem comes when trying to access from outside via wifi. 

Configuring networks can be quite confusing when doing it for the first time. It's a good decision to **not** run your own DHCP server. Speaking from experience. :smile: 

I understand your network layout looks similar to this!?

[image]

The point where you seem to have trouble is the connection between your AP and the (old) router, correct? Did you manage to connect *any* devices that are on the different networks?

The AP seems to give out IP Addresses like `10.55.*.*` or `10.240.*.*`, but your ATEM has the address `192.168.10.240`. Did you configure the **routing tables** to connect those two networks? 

How you do this is highly depended on the devices you use. What you basically need to achieve is to tell your AP "if you get packages for an IP address that matches the pattern `192.168.178.*`, send it to the router instead". And your Router needs to know "if you get packages for an IP address that matches `10.55.*.*` or `10.240.*.*`, send it to the access point".

I hope this helps you further and the information was not too basic or high-level. **Routing table** and the name of your AP or router is what you might want to search for.

If it does not help, feel free to check back.

pwright2b, 2021-05-24:
Routing Tables, oh dear.  I have never ventured that deep into the jungle before.  But if that is what I need, I shall venture forth.  I do appreciate your guidance.  Yes, you visualize the network layout properly.   The router is a Netgear unmanaged switch, so as far as I can tell it doesn't change anything passing thru it.   I appreciate your confirming that the address I am trying to reach is  http://192.168.10.55:3000.  I believe I have the access point configured to give out 192.168.10.x addresses but I will do what I can to double check that.

-----Paul-----

pwright2b, 2021-05-24:
OK, I thrashed around a good bit today.  Made enough changes to the router setup that I feared I would never get it going again.  Network configuration is truly a blend of witchcraft and black magic.  Anyhow, in the end it seems that my technology has been fooling me and it has been working all the time.  At any rate I can log in via wifi to 198.164.10.55:3000 using a laptop browser and the remote Tally function works just fine.   But, for the life of me I cannot get a phone based browser to successfully log in.  I have tried an android and an iPhone (old).  They both attach to the network and accept the URL and appear to begin to load the page but then stop, with nothing rendered.  No error codes, just stop.

Anyhow, I am ready to proceed to the next step - making the actual Tallys.  There will probably be more messages.

-----Paul-----

pwright2b, 2021-05-30:
As I mentioned before, the hub and web tally's work fine.  My worship leader was able to log in with his iPhone, so I don't know why my phones won't hook up.  But I have been working of the wifi tallys.

I bought these (https://smile.amazon.com/gp/product/B07L8W9SP3/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1) and this (https://smile.amazon.com/gp/product/B07BKPS3G1/ref=ppx_yo_dt_b_asin_title_o02_s01?ie=UTF8&psc=1)

I had played with programming an Arduino years ago, so the programming process was not COMPLETELY foreign, though I struggled with it.  I think I succeeded.  The LED strip is a 3 wire as illustrated in the Reference rather than a  4 wire like in the Starting instructions.

I have yet to see anything from the LED strip.  On the 8266 board itself there is some reaction which is undescribed in the instructions.  So I thought I would ask if this is meaningful, either in a good or bad way.

There are two blue LEDs on the board.  One is on the upper right corner.  The other is just to the left of the USB port.  Without the LED strip attached, the blue light in the corner is on solid and the blue light next to the USB is blinking about once a second.  

Attaching the LED strip by the diagram in tally.html generally causes both blue LEDs to go out.  Though not always.  I haven't discerned a pattern.  Attaching a simple single color LED between 3V3 and D4 causes the LED to light, the flashing blue LED to go out and the steady LED remains on.

I'm hoping some of this means something.  I don't have a way to test the LED strip since it apparently needs a modulated signal to produce any light at all.  I suppose I should have ordered a controller with it.

-----Paul-----

pwright2b, 2021-05-30:
And so now I have ordered a controller.  At least I will be able to determine whether the LED strip is defective.

czenker, 2021-05-30:
Great to hear you made it one step further.

> I have yet to see anything from the LED strip. On the 8266 board itself there is some reaction which is undescribed in the instructions. So I thought I would ask if this is meaningful, either in a good or bad way.

It is not described in detail, because not all boards have this LEDs. But it seems you got the same build that I have.

Flashing is usually a good sign - it means the microcontroller is doing *something*. It seems your flashing process has worked.

Consider the LED in the top red corner as an indicator that the board is powered. The other LEDs flash pattern is described in the [Troubleshooting Guide](https://wifi-tally.github.io/troubleshooting.html#the-tally-slowly-blinks-blue-for-an-extended-amount-of-time). You should find the pattern you are seeing and some possible causes there.

> I'm hoping some of this means something. I don't have a way to test the LED strip since it apparently needs a modulated signal to produce any light at all. I suppose I should have ordered a controller with it.

Nah. You are good to go. What you are having is a WS2812 compatible LED strip – the microcontroller can drive that too. See the detailed [tally documentation](https://wifi-tally.github.io/tally.html#using-ws2812-neopixel-etc) on how to set it up. It's just a configuration and a little different wiring.

> Attaching the LED strip by the diagram in tally.html generally causes both blue LEDs to go out.

This sounds a bit concerning. This usually means that the controller stopped working, because of a short circuit or too high power drain. This does not do permanent damage in most cases, but it is one of the things that you should not do too often. And please take extra notice of this warning in the documentation: "Do not connect more than 5 LEDs to the board!" This might definitely drain too much current if you connected all 144 LEDs.

pwright2b, 2021-05-30:
Thank you.  I have been doing my testing at home where there is not the wifi signal the Tally is looking for.  I will take it to church tomorrow and see if I get something new.  I have actually dropped back from 5 to 3 LEDs on the strip, just in case of excessive power draw.  The blue LED flash pattern - I was thinking the Troubleshooting Guide was talking about the WS2812 LEDs.  This is actually encouraging.

-----Paul-----

pwright2b, 2021-05-31:
If I need to correct the tally-settings.ini that I uploaded, can I just re-upload the corrected file or do I need to erase the entire chip and start over?

Hope you are having a pleasant Memorial Day.

pwright2b, 2021-06-01:
Was able to answer my own question.  While examining the tally-settings.ini examples in the instructions I noticed that I had messed up the hub.ip in my upload.  I had 12.168 instead of 192.168.  With that corrected, the tally connected to the hub near instantly, was recognized, and could be assigned to a camera.  Only thing it couldn't do was light up the LED strip.   Nothing shows there.  I tried moving around the connections.  No change.

On Thursday, Amazon is to deliver a controller for the strip.  Then I should be able to prove whether I destroyed the LEDs while trying to solder to those tiny pads or what.  

I kind of wish there was a simpler version that would simply light up discrete LEDs.  Would seem simpler to diagnose.  But I look forward to exploring the mysteries of this one as they present themselves.

-----Paul-----

pwright2b, 2021-06-01:
**IT WORKS!!**  First I decided to guess that the reference to Common Cathode might mean some discrete LEDs with the cathodes tied together.  Then I had to look up how to determine which was the cathode.  Then I spent an hour fruitlessly testing.  Then I reread the instructions and finally noticed that it said to connect to GND rather than 3v3.  And then I tested some different connection points and finally got an LED to turn on ... and off.  And then another LED at a different connection point turned on and off.  And now it tells me whether the camera is on preview or live.  

Whahoo!  Thank you.  This is great.  Now if I can just package it in a moderately attractive package ....

*****Not your problem. *****

But maybe you could add a few more explanatory words to the section about Common Cathode.   1.  That it refers (or can refer) to discrete LEDs, and 2.  That they will use the same pins (D1, D2, D3) as the Build and Setup a Wifi Tally page EXCEPT using GND instead of 3v3.

-----Paul-----

czenker, 2021-06-01:
Awesome!

It's always tricky to get documentation right to cater different audiences, but you are right there could be more details or an example schematic on your setup. Feel free to create a pull request on the documentation with the information that might have helped you. In the future I hope to have guides for different setups – with one recommended setup.

Just one thing to add: Did you add any resistors to the LEDs? When using discrete LEDs you should always add a resistor between the pin D1, D2, D3 and the respective pins on the LED. Otherwise you might risk burning the LED. It usually does not happen quickly, but reliability is key for a tally light. :wink: `150Ohm` for red and `100Ohm` for blue and green are common values to use.

czenker, 2021-06-02:
> Does the value for operator.ws2812 need any adjustment?

No. If you don't use it, it does not matter what you set. You could set it to `0`, but you would not see any performance difference anyways.

pwright2b, 2021-06-07:
Your comment on 'create a pull request' reminded me how little I know about the github environment. And then I got a note about renaming to vtally.  So I looked around in history.

.... My god you have poured a lot of work into this.  And I am so grateful for your support.  

So far I have built a Mk 1 [(cardboard box size of an old cigarette case), ugly as home made sin but works], Mk 1 Mod 1 (smaller plastic case), and  a MK 2 model where the lights are on a cord independent of the case.  Trying to build a second Mk 2 but my clumsy fingers are in the way.

So far have not succeeded in incorporating WS2812, only individual LEDs, but they work fine and serve our purpose.

I don't want to violate githubs standards but would be interested in trying the 'pull request' and adding some commentary.  Is there a style guide or whatever I should peruse?

----Paul------

czenker, 2021-06-13:
> I don't want to violate githubs standards but would be interested in trying the 'pull request' and adding some commentary.

There is not much to worry about. 😄 Everyone of started with their first pull request at some point. The good thing is: You can't really break anything. So don't panic. I can have a look before your changes are integrated.

> Is there a style guide or whatever I should peruse?

There is not really a styleguide. The documentation is generated by some files in the [documentation](https://github.com/wifi-tally/wifi-tally/tree/main/documentation/docs) directory. The file format is markdown (the same markup language you can use in the github comments) plus a few plugins for variable substitution and info boxes. You can edit it with any text editor or even with the browser editor on Github. The syntax should be pretty obvious, if you compare the markdown files with the rendered documentation.

If you want to see your result before submitting to github this should be really easy on a linux machine (you are working on the Eee Pc, right?). [Brief documentation is here](https://github.com/wifi-tally/wifi-tally/blob/main/documentation/Readme.md), but this should work for you on Debian:

````bash
# this should be installed by default, but to be sure
sudo apt-get install python3

cd documentation/
python3 -m pip install -r requirements.txt --user

mkdocs serve
````

and open your browser on http://127.0.0.1:8000 It should update the documentation shown everytime you save your changes.

czenker, 2021-12-28:
Discussion went stale. Closing.

closed 2021-12-28 by czenker