wifi-tally/wifi-tally #72: Help Needed: Test v0.5.0 before official release

czenker, 2021-05-15:
Hey everyone. 

I'm almost ready to publish the next big release. But as the way of installation has changed (hopefully to be even easier now), I want **as many people as possible to test it on their hardware** to iron out any incosistencies or bugs that might come up.

The new release includes multiple changes that I am really exited about:

1. We have a logo. And a new name :tada: 
    [image]
2. @JWandscheer added support for Roland VR-50HD and V-8HD :clap: 
3. It is now possible to modify `tally-settings.ini` and update the code from the Hub via USB.

The latter two changes also required the change in the installation procedure.

So please give it a try. I post the new installation instruction right here as the first comment. If it works for you or you run into issues, please use this issue to report.

----

## Windows

### 1. Install Node.js ≥12

[Install Node.js](https://nodejs.org/en/download/package-manager/#windows).

Verify your installation

    $ node -v
    v14.15.4
    
    $ npm -v
    6.14.10

### 2. Install vTally

This command will install vTally globally for all users. Run

    npm install -g vtally:0.5.0-alpha2

### 3. Run vTally

vTally is installed. Run it by calling

    C:\Users\account_name\AppData\Roaming\npm\node_modules\vtally

## Linux

### 1. Install Node.js ≥12

[Install Node.js](https://nodejs.org/en/download/package-manager/) via your distribution's package manager.
It is likely you already have it installed.

Verify your installation

    $ node -v
    v14.15.4
    
    $ npm -v
    6.14.10

### 2. Install vTally

This command will install vTally globally for all users. Run

    sudo npm install -g vtally:0.5.0-alpha2 --unsafe-perm

!!! info
    The words `sudo` or `--unsafe-perm` might concern you. The line above is the typical way to install global npm
    packages with precompiled node extensions. You can read below for an alternative installation for one user.

### 3. Run vTally

vTally is installed. Run it by calling

    vtally

### Alternative installation for one user

You can install vTally – for one user only – through the following steps:

    npm config set prefix '~/.local/'
    echo 'export PATH=~/.local/bin/:$PATH' >> ~/.bashrc
    source ~/.bashrc
    npm install -g vtally:0.5.0-alpha2

Amenasme, 2021-05-16:
I will try... seems very difficult for not a programmer (me).
Where do I have to insert all these codes?

czenker, 2021-05-20:
> Where do I have to insert all these codes?

You run on Windows, right?

Just open the command line (windows + type "cmd" + enter). Don't fear the command line – it's friendly most of the time. :laughing:

Amenasme, 2021-05-26:
I tried and I stopped with errors at the beginning. I can't do that, it's too difficult for me.  Can you do a step by step video of the installation process? Maybe in the future a  *.exe  file will help people like me (not a programmer).

czenker, 2021-05-29:
> I tried and I stopped with errors at the beginning. I can't do that, it's too difficult for me. Can you do a step by step video of the installation process? Maybe in the future a *.exe file will help people like me (not a programmer).

I uploaded a video how the initial installation should ideally look like on Windows: https://vimeo.com/556543349

Where exactly do the errors occur? What do the errors say?

Amenasme, 2021-06-01:
Thanks for the video, it is useful but I got this error anyway. I installed node.js as in your video. What I have to do?
Thanks

[image]

Amenasme, 2021-06-01:
I have a 64bit windows OS

[image]

czenker, 2021-06-01:
> Thanks for the video, it is useful but I got this error anyway. I installed node.js as in your video. What I have to do?
> Thanks

Hey. 

I guess you have downloaded the 32bit node installer by accident. Remove it and install the 64bit version. The latest current version is https://nodejs.org/dist/v14.17.0/node-v14.17.0-x64.msi

Then try again.

Amenasme, 2021-06-01:
> I guess you have downloaded the 32bit node installer by accident

Yes, I did. I don't know why, and I didn't noticed it. 

Now I removed it and installed the version that you linked to me. I ran cmd and wrote:

npm install -g vtally@0.5.0-alpha2

and got all these errors

`
C:\Users\Salvatore>npm install -g vtally@0.5.0-alpha2
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\es-abstract-71d0f476\2016\CreateDataProperty.js'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\es-abstract-71d0f476\2017\CreateDataProperty.js'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\es-abstract-71d0f476\2018\CreateDataProperty.js'
npm WARN tar ENOENT: no such file or directory, lstat 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\vtally-f2aa0d13\src\frontend\static'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\atem-connection-3fd4e63b\dist\commands\TallyBySourceCommand.d.ts'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\es-abstract-71d0f476\2016\CreateDataPropertyOrThrow.js'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\atem-connection-3fd4e63b\dist\commands\TallyBySourceCommand.js'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\es-abstract-71d0f476\2017\CreateDataPropertyOrThrow.js'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\atem-connection-3fd4e63b\dist\commands\TallyBySourceCommand.js.map'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\es-abstract-71d0f476\2018\CreateDataPropertyOrThrow.js'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\atem-connection-3fd4e63b\dist\commands\TimeCommand.d.ts'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\atem-connection-3fd4e63b\dist\commands\TimeCommand.js'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\atem-connection-3fd4e63b\dist\commands\TimeCommand.js.map'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\atem-connection-3fd4e63b\dist\commands\index.d.ts'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\es-abstract-71d0f476\2016\CreateHTML.js'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\atem-connection-3fd4e63b\dist\commands\index.js'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\es-abstract-71d0f476\2017\CreateHTML.js'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\atem-connection-3fd4e63b\dist\commands\index.js.map'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\es-abstract-71d0f476\2018\CreateHTML.js'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\midi-a3c0c2cc\test\input\virtual-input-test.js'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\atem-connection-3fd4e63b\dist\dataTransfer\dataLock.d.ts'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\atem-connection-3fd4e63b\dist\dataTransfer\dataLock.js'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\atem-connection-3fd4e63b\dist\dataTransfer\dataLock.js.map'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\es-abstract-71d0f476\2016\CreateIterResultObject.js'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\atem-connection-3fd4e63b\dist\dataTransfer\dataTransfer.d.ts'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\es-abstract-71d0f476\2017\CreateIterResultObject.js'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\atem-connection-3fd4e63b\dist\dataTransfer\dataTransfer.js'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\es-abstract-71d0f476\2018\CreateIterResultObject.js'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\atem-connection-3fd4e63b\dist\dataTransfer\dataTransfer.js.map'
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\atem-connection-3fd4e63b\dist\dataTransfer\dataTransferAudio.d.ts'
[... cut ...]
npm WARN tar ENOENT: no such file or directory, open 'C:\Users\Salvatore\AppData\Roaming\npm\node_modules\.staging\engine.io-3e23009d\lib\transports\websocket.js'
npm ERR! code ENOENT
npm ERR! syscall spawn git
npm ERR! path git
npm ERR! errno -4058
npm ERR! enoent Error while executing:
npm ERR! enoent undefined ls-remote -h -t ssh://git@github.com/wifi-tally/NodeMCU-Tool.git
npm ERR! enoent
npm ERR! enoent
npm ERR! enoent spawn git ENOENT
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

`

Amenasme, 2021-06-01:
node.js seems correctly installed

[image]

stevsimo, 2021-06-02:
Zenki ich hab die selben Fehler!

czenker, 2021-06-03:
@stevsimo @Amenasme Sorry for the trouble. :confused: I was hoping it worked with less effort. 

I created an exe file and I hope it works better. Updates in the initial post.

It is a bit rough on the edges right now (tons of security warnings) and I think it should at least have a tray icon as @ruudboon suggested in #9.

Also: If anyone with a sense for adventure could give the MacOS build a test drive – that would be awesome.

JWandscheer, 2021-06-04:
MacOS seems to work without any issues.
After installing, you need to open the Application by right click -> open, otherwise MacOS will give a security warning without the option to run the app nevertheless.

czenker, 2021-06-05:
Thanks for all your help. I have just released [v0.5.0](https://github.com/wifi-tally/wifi-tally/releases/tag/0.5.0) Thanks to everyone who made it happen! :tada: :fireworks:

closed 2021-06-05 by czenker