wifi-tally/wifi-tally #77: Increase maximum number of ws2812 leds to at least 30

AndreBott83, 2021-07-08:
Could you change the my-settings.lua file to support higher numbers of ws2812 leds for Operator and Stage lights?
E.g. increase the input checks to 30

czenker, 2021-12-29:
Sorry for the late response.

What is the use case here?

Limiting the number of LEDs is mostly done for two reasons:

* Writing data for WS2812 takes time. The more LEDs you have, the longer it takes, which might lead to laggy output. I'm not sure about the NodeMCU-internals, but the `wifi` and `ws2812` module should both work with interrupts. Just from experience I think it is better to reduce the workload inside interrups as much as possible. Otherwise this could cause dropped UDP packages or flickering LEDs
* NodeMCU boards have a limit to total current drain of about `250mA`. Having too many LEDs connected might potentially damage the board.

If there is a need for an "expert option" to support more LEDS, there might be some way to support it. But you should know what you are doing and can not just follow the documentation.

AndreBott83, 2021-12-30:
Hi, 
don't worry about the delayed response and thx for the extensive answer.

The use case was to make use of  WS2812 LED rings as a tally signal. E.g. the NeoPixel ring (https://www.amazon.de/gp/aw/d/B01B35DDTU)
The 24 LEDs are permanently attached and it looks a lot nicer if all of the leds are on.

The power supply is directly supplied to the soldering terminals and not taken through the NodeMCU board outputs to not toast the internals or  trip a fuse.

Latency and connection looks fine. I did also not notice a flickering of the leds with the number set to 24+2 in my test.

But I agree, maybe that extension should be behind some expert Modus to be on the safe side

sinnoken, 2024-10-10:
Can it be increased to 64, for example using 8x8 RGB Shield？
https://www.wemos.cc/en/latest/d1_mini_shield/8x8_rgb.html