wifi-tally/wifi-tally #78: Problem with Atem connection

IbroZe, 2021-08-01:
version 0.5.0 tally hub for windows cannot be connected to Atem mixer.  I tried from another computer but without success.  Version 0.3.0 connects to atem mini pro and atem production studio without any problems.

sisensee, 2021-08-22:
I can confirm same problem for a Mac. Two Macs, two OS versions, tally's connect and update code accordingly, but no connection to ATEM.

Numelle, 2021-11-01:
Same Problem here. Scenarios tested:
Scenario 1:
- ATEM Mini Pro
- Mac (Intel) BigSur  11.6 (normal firewall configuration - no special changes)
- vTally Hub 0.5.0
- 1 Tally connected
- Port 9910 from HUB Device to ATEM Mini Pro verified and open
- No other network connections to ATEM than the vTally Hub
Result:
NO Connection established
Scenario 2:
- ATEM Mini Pro
- Mac (Intel) BigSur  11.6 fresh installation without AV or FW tooling
- vTally Hub 0.5.0
- 1 Tally connected
- Port 9910 from HUB Device to ATEM Mini Pro verified and open
- No other network connections to ATEM than the vTally Hub
- Clean separate network on a dedicated router.
Result:
NO Connection established
Hope this helps solving the issue for 0.5.0.
Thanks! ;-)

Numelle, 2021-11-09:
Hey @czenker @kubikrubik21  I was wondering if there is an update to this issue?
Are there any others who have this issue with current version and want to connect with ATEM?
Is there any additional information needed from my side?
Thank you very much ;-)

czenker, 2021-11-13:
Finally got around to getting some logs

````
[2021-11-13 15:17:16.309] [error] (node:10256) UnhandledPromiseRejectionWarning: Timeout when trying to restart
[2021-11-13 15:17:16.310] [error] (node:10256) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 5691)
[2021-11-13 15:17:16.345] [error] (node:10256) UnhandledPromiseRejectionWarning: Timeout when trying to restart
[2021-11-13 15:17:16.345] [error] (node:10256) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 5692)
[2021-11-13 15:17:16.357] [error] (node:10256) UnhandledPromiseRejectionWarning: Timeout when trying to restart
[2021-11-13 15:17:16.357] [error] (node:10256) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 5693)
[2021-11-13 15:17:16.362] [error] Worker Thread error Error: Cannot find module 'C:\Users\shark\AppData\Local\Temp\20rnAXhmnuYhkEx6FPpDWyUy8uD\resources\app.asar\node_modules\threadedclass\dist\child-process\threadedclass-worker.js'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:887:15)
    at Function.Module._load (internal/modules/cjs/loader.js:732:27)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:72:12)
    at MessagePort.<anonymous> (internal/main/worker_thread.js:174:24)
    at MessagePort.[nodejs.internal.kHybridDispatch] (internal/event_target.js:354:41)
    at MessagePort.exports.emitMessage (internal/per_context/messageport.js:18:26)
````
This is obviously something [atem-connection](https://www.npmjs.com/package/atem-connection) internal, that does not like to run inside electron. Unfortunately I do not expect there to be a quick solution.

For now the best way of action would be to [install vTally from npm](https://wifi-tally.github.io/download.html#run-from-npm). You miss the tray icon, but it works with Atem.

czenker, 2021-12-28:
The issue is mentioned in https://github.com/nrkno/tv-automation-atem-connection/issues/106 Updating atem-connection to version 3+ could help.

Numelle, 2021-12-28:
> Antworten an: wifi-tally/wifi-tally ***@***.***>
>
>
>
> The issue is mentioned in nrkno/tv-automation-atem-connection#106 Updating atem-connection to version 3+ could help.

Hi there,

 

thanks for your efforts, but to be really honest I do not know which part to update

to fix the issue.

With regards to the MAC package could you give me a quick hint how to start?

 

Or is there a plan to do an update in the package accordingly?

 

Thanks

Andreas

czenker, 2021-12-28:
Sorry for closing this issue too early. I just released a new build [v0.5.1-rc1](https://github.com/wifi-tally/wifi-tally/releases/tag/0.5.1-rc1). This seems to fix the issue for me. Does it work for you, too, @Numelle? If so, I would do a proper release.

Numelle, 2021-12-28:
Hi Christian, will test that accordingly. Thanks for that. You will get asap Feedback.

czenker, 2022-01-21:
This should be fixed with [v0.5.1](https://github.com/wifi-tally/wifi-tally/releases/tag/0.5.1).

commit 3b292014 (2021-12-28): Update atem-connection to dev version 3.0 hoping it might fix #78
closed 2022-01-21 by czenker