wifi-tally/wifi-tally #84: Lights off when video mixer disconnected

wde, 2021-10-30:
Hi there, I love this project! When using OBS (haven't tested with ATEM yet) and we shut down OBS the last known status remains and then if we close vTally Hub the units default to the double-flash blue connection warning.

We are putting these onto PTZ cameras that also double as security cameras when the ATEM is off.

Is there a way to get the LEDs to sleep when either the hub or ATEM is not connected?

I get that there needs to be a signal that the tally light is ready, maybe after 10 minutes or so it can switch to a tiny dim flicker every 20 seconds or so. Mostly in order to avoid it distracting people in the theatre. Or even a fade in/out dimly.

Many thanks for a great idea.
Cheers,
Sean

czenker, 2021-11-13:
Hey and welcome, Sean.

> we shut down OBS the last known status remains and then if we close vTally Hub the units default to the double-flash blue connection warning.

That sounds like a bug. It should start flashing with some pattern when OBS disconnects. But I understand that is not your main concern. 😄 

> Is there a way to get the LEDs to sleep when either the hub or ATEM is not connected?

I guess, you are operating your cameras remote then? So I guess, you would not need a dedicated light for the operator. How about you only connect the Stage Lights, but not the Operator Lights? ([documentation here](https://wifi-tally.github.io/getting-started/setup-wifi-tally.html#separate-operator-and-stage-light)) The Stage Light never shows _any_ status other the `Preview` and `Program`. When using WS2812B strips, it should be possible to set this to have the same effect:

````ini
operator.ws2812 = 0
````

This would prevent the tally from ever flashing blue.

Does this help?
Christian.

wde, 2021-11-23:
Thanks Christian, That is a great idea. I am using WS2812Bs.

I might have a look at the code and play with it locally. If I come up with any brilliant ideas I'll submit them. Still love the project as broadcasting becomes more ubiquitous most of us can't afford a $6,000 tally system like the TV studios of the "olden days".

Cheers,
Sean

closed 2021-11-23 by wde