wifi-tally/wifi-tally #85: vMix Channels

zuperzuche, 2021-11-02:
Before building wifi modules I decided to test web tallies with vMix. As far as I understand only 8 channels are supported and they correspond to inputs 1-8. I was wondering if it is possible to have more channels?

czenker, 2021-11-13:
Hey @zuperzuche and welcome.

If you select vMix as your video mixer and the connection is successful, vTally Hub should show the correct number of channels and their channel names. If not, let me know - this would be a bug.

The 8 channels is just a default if no video mixer is selected yet.

MartLeib, 2022-01-20:
It seems that I have a similar issue, If I select vMix and enter vMix Web Interface IP and port, then it connects to vMix for a second or sometimes longer, but never shows input names, only Unpatched and Channels 1-8.

Log is attached.
[main.log](https://github.com/wifi-tally/wifi-tally/files/7903895/main.log)
.

czenker, 2022-01-21:
Hey @MartLeib.

Thanks for adding the log. This really helped figuring out what's wrong: It is the port. It needs to be port `8099` of the [TCP API](https://www.vmix.com/help24/index.htm?TCPAPI.html), *not* the web UI.

If this fixes the issue, I would leave this ticket open and change its title, because the log should not say, it is connected to the mixer as this is obviously not the case.

MartLeib, 2022-01-22:
Hei @czenker when I changed 8088 to 8099 it worked right away. No need to restart or anything.
I would recommend not showing port selection for vMix at all because the port 8099 for TCP API is fixed and users can not change it. Users can change only the Web Controller port.

czenker, 2022-02-03:
There are some – admittedly rare – cases where the port might be changed when doing some network magic. So i would rather keep this logic in. I hope that the warning and logs added in the commits above will prevent users from changing the port unintentionally.