wifi-tally/wifi-tally #99: Multiple sources.

mndmr, 2022-03-04:
I have two sources in which a  camera is on. Is it possible to patch to multiple sources?

Furchendackel, 2023-04-25:
+1

Sometimes you have multiple scenes in OBS where the same camera is active (split screen or PiP). Would be great and useful for such circumstances.

czenker, 2023-04-27:
Hey @Furchendackel 

> Sometimes you have multiple scenes in OBS where the same camera is active (split screen or PiP). Would be great and useful for such circumstances.

This is already possible with OBS. Create a scene that just contains your camera image full screen. You can use this scene as source in a different scene for PiP or split screens. Configure your tally to track the full screen scene and it should also activate anywhere this scene is embedded into a different scene. I have not tested it in the last year, but it worked in the past.