wifi-tally/wifi-tally #105: add battery status

davide92br, 2022-04-15:
I'm building a tally with wemos d1 mini with integrated battery and it would be nice to include a function that checks the battery status from the web server so that I can monitor the charge level.

GwiezdnySzeryf, 2022-05-24:
It would indeed be great function to have!

czenker, 2022-05-24:
Sounds like a great idea. I am just not sure how this would technically work. With some "stupid" battery the best you could do is monitor the voltage and guess how much capacity is left. But with all the different batteries and chips out there, this seems to be closer to reading tea leaves. 

Does anyone have practical experience how to implement this savely? Or any implementation in a similar project?