wifi-tally/wifi-tally #107: Change Data pin

Stephan-Kl, 2022-04-30:
Hello,
i have found your solution for a tally light for vMix and its super easy to set up.
I already have build the Tallys with a different software for the atem mini and want to switch to vMix.

The problem is that I soldiers the Neopixel on pin D7.
Is it possible to change pin D4 to D7 in the settings?

Thank you and kind regards
Stephan

czenker, 2022-05-01:
Hi Stephan.

I had to look into it, but unfortunately it does not seem to be possible. 

The [underlying library only supports `D4` as UART Pinout](https://nodemcu.readthedocs.io/en/release/modules/ws2812/#ws2812init). The only other option would be to use `TX`, but this would essentially break data over USB. 

This is probably something that they should support. But before I open an issue there: Are you sure it is `D7` you are using? According to the pinout `TXD2` is on `D8`.

Stephan-Kl, 2022-05-01:
Hi,
thank you for the fast reply. Unfortunately it is D7 :D so I have to desolder the cables, its niche that difficult but I thought I can change it the easy way ;)