wifi-tally/wifi-tally #115: Vtally on Raspberry pi 4

gvanos, 2022-10-02:
I'm not sure whether this is an issue or error on my part.

I'm trying to run vtally on a raspberry py 4 with with 64 bit Debian DNU (bullseye) installed.
Installed nodejs and was able to make a build as there is no vtally software for ARM64-architecture.

The build starts but when I open localhost:3000 there is a message telling me that the hub is disconnected. ALso, when I try to do CONFIGURATION on the browser page, the Video Mixer and Tally Defaults areas only show a rotating circle.

What am I missing? Any ideas?

P.s. I have a hub running on a laptop and we are very pleased with that. The laptop, however, is also used for other things so that made me think to have a vtally hub on a RPi.

gvanos, 2022-10-02:
I missed something! I needed to start the backend in order for the whole thing to work.

It works fine now!

closed 2023-04-27 by czenker

manuugonz, 2024-03-18:
Would you be so kind to upload your raspberry pi already working with this document? I've tried anyway to get it working and when I run npm config set prefix '~/.local/'
echo 'export PATH=~/.local/bin/:$PATH' >> ~/.bashrc
source ~/.bashrc
npm install -g vtally:0.5.1

I get an error and I can't make any progress....
I have Raspberry pi 4 model B