wifi-tally/wifi-tally #131: OBS Websocket 5 support

sky7176, 2023-06-05:
Good stuf but with a new version of websocket, from OBS 29 and above, this plug-in no longer works.
Are you going to make the necessary changes to get everything working again?

Thank you
Giuseppe

nicolopozzato, 2024-02-12:
So I'm here to ask, I see this project yesterday, I'm now interested on making this for my local church (in some event we use 1 or 2 remote camera, like a smartphone with ndi camera on a smratphone stabilizer) the problem is the comunication from the managment desk to the operator... i really like this project to have a small esp and 2-3 led to notify the camera operator if it's live or in preview. But we use OBS Studio so I don't know now what to do

sergal-IRL, 2024-04-03:
The current version of wifi-tally does not work with OBS version 28.x and above due to the inclusion of obs-websockets 5.0 in that version onwards. 

A temporarily fix is to disable the 5.0 websocket server, and install the 4.9.1-compat version from OBS themselves (https://github.com/obsproject/obs-websocket/releases)

iilychy, 2024-12-21:
> The current version of wifi-tally does not work with OBS version 28.x and above due to the inclusion of obs-websockets 5.0 in that version onwards.
>
> A temporarily fix is to disable the 5.0 websocket [...]

Seems that 4.9.1-compat couldn't be installed and work properly on Linux, at least on my Ubuntu 22.04 with OBS 30.
I've seen same cases happened too. 
https://github.com/obsproject/obs-websocket/issues/1166#issuecomment-2151523584

And contributers of obs-websocket are not gonna fix this bug for now.

So vtally only supports websocket 4, but websocket 4 compat can't be installed on linux. Seems to be dead end.

Is there a way to use vtally without rollback to obs 27 on Linux? Thanks.